bambulab/BambuStudio #5226: The pruge line on V1.10 is causing issues. Please go back to the old purge line

thrutheframe, 2024-11-13:
The new purge line on top of the calibration lines is a terrible idea. 

The purpose of a purge line is to clean the nozzle before the print. If you purge on an existing print line, it's going to pick up any stray filament, creating a blob at the tip of the nozzle and then it gets transferred on the print, creating a bump. It is especially prominent when printing petg. and I am using bambu petg-HF. I had 1 fail prints and had to stop 3 more because of it. 

Please go back to the old purge line located on the top right-hand corner.

OzDruiD, 2024-11-13:
God no, not the old one, but your point is valid, not creating a new problem either.
I will try and replicate as I have not seen this even in BETA, will post steps here as I am yet to have this problem, but if it does , yes its an issue that neeeds a fix.

thrutheframe, 2024-11-13:
One way is to do a shorter purge followed by a wipe on the build plate. It was a custom start gcode that I shared in the past.

In the code below, I used the right edge... but I would recommend having it at the rear of the build plate. 

;===== Nozzle load line sequence ===============
G1 E{-retraction_length[initial_extruder]} F1800
G0 Z5 F300          ;move heatbed down by 5mm
G0 X254.5 F6000     ;move toolhead to the right
G0 Y50 F6000        ;move toolhead 50mm on Y axis
G0 Z1.0             ;move heatbed to 1.0mm to allow oozing. 
G1 E{retraction_length[initial_extruder]} F1800
M109 S[nozzle_temperature_initial_layer]
M400
G0 Z0.2             ;lower nozzle to printing height. 
G1 E1 F300          ;extrude 1mm of filament at 5mm/s
G0 X253 
G1 Y175 E6.4 F{outer_wall_volumetric_speed/(0.3*0.5) * 60} 
G0 X251.5
G1 Y50 E6.4

;===== Quick wipe sequence ===============
G0 Y48 Z1.0 F6000   ;1mm lift and move 2mm on Y
G0 Z0.0             ;press nozzle onto bed
G0 Y44 F3000        ;start of 4 wipes movement
G0 Y48 F3000
G0 Y44 F3000
G0 Y48 F3000
G0 Y44 F3000
G0 Y48 F3000
G0 Y44 F3000        ;end of 4 wipes movement
G0 Z0.4             ;raise nozzle up to prevent scratching bed

tangketan, 2024-11-14:
The purge line is not on the top of an existing line. It's just near one, so you can remove all the lines once.
We have changed the purge line for months internally and done quite a lot of tests without observing any issue. Would you give more details such as photos or videos of the failure, your printer model, filament model, etc.?

thrutheframe, 2024-11-14:
Will take some videos regarding this and share the observation. I am also waiting for the new bambu supertack plates to arrive to try and see if the issue happens too.

pureshem1, 2024-11-24:
Another example of the 3rd line pulling into a print after catching on the initial calibration line blob.  Near mishap, print salvageable.
[image]