bambulab/BambuStudio #5230: Painted Support Blocker not 'Blocking' tree supports

CraigHutchinson, 2024-11-13:
### Bambu Studio Version

1.10.0

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

1. Paint 'Support Blocker' on areas of model for which we don't want supports (in this case they are irremovable/unecessary and we can accept loss of quality etc) e.g. 
[image]
2. Slice model with tree supports
3. Observe supports are made for blocked areas

### Actual results

Supports are generated for 'blocked suport' regions of the model so supports cannot be removed. In this reproducer case we don't want supports to go inside the central area of the mouth and tightly between the teeth.
[image]

[image]

### Expected results

Support should not be generated for the internal region which was painted with blocker! Painted support blockers shall block the support generation here.

Adding a sphere blocker didn't work either!

### Project file & Debug log uploads

[Testa di Drago_B.zip](https://github.com/user-attachments/files/17732170/Testa.di.Drago_B.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-11-14:
Painted support blockers only prevent generating new supports at the painted position. Other supports can still pass around. If you look carefully, there are NOT support starting from the painted area, which is the expected result.
Would you explain your original need, so maybe I can figure out a way to meet your need?

CraigHutchinson, 2024-11-14:
Thanks for response. I was mistaken as I thought this support looks to be for the painted area? (As there are no 'support interface' being generated for any of the supports its hard to be certain what this is otherwise here for)
[image]

The issue is to prevent supports entering inside the mouth section and between/behind the teeth.

tangketan, 2024-11-15:
I see. The interfaces are not always generated using organic support, unless you are using "Support for PLA" as the interface material. 
"Preventing supports entering some area" can be useful, but difficult to implement. I'll keep it in mind.