bambulab/BambuStudio #5231: Bambu Studio v1.10 initially fails to show filament path from AMS in UI

Tumsk, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 24H2

### Additional system information

CPU: Intel(R) Core(TM) i7-8700 CPU
RAM: 32.0 GB
OS Build: 26100.2314

### Printer

Bambu Lab P1S

### How to reproduce

1. Slice and Print project with AMS in use.
2. Wait for the Printer to load filament from AMS.
3. Observe AMS UI in Bambu Studio.
4. Only tool head and extruder show filament colour from AMS in UI in Bambu Studio.
5. Wait for some time (approx. 1 hour / 10% of print / c. 20 layers).
6. Observe AMS UI in Bambu Studio.
7. Full filament path is now shown in AMS UI in Bambu Studio.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17732030/log.zip)
[Project.zip] File size too big

### Checklist of files to include

- [X] Log file
- [ ] Project file

an565, 2024-11-16:
Bambu Studio Mac 1.10.1.50 seems to have this bug as well.

walterwongbbl, 2024-11-21:
@Tumsk @an565 
Thank you very much, we will try to fix this issue.