bambulab/BambuStudio #5232: Floating layers - aka layers missing when slicing thin/narrow features using `Archane`

CraigHutchinson, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

1. Select `Archane` slicer
2. Zoom in on the 'thin' teeth and obseve some gaps where a layer isn't printed for the feature

### Actual results

There are floating layers in the model for thin points on numerous teeth on this model.
[image]

Normaly archane shdouln't face this issue but it appears to fail on specific features of this model.

### Expected results

There sholdn't be floating layers

### Project file & Debug log uploads

[Testa di Drago_B.zip](https://github.com/user-attachments/files/17732481/Testa.di.Drago_B.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

XunZhangBambu, 2024-11-14:
@CraigHutchinson  
Hello, even with the default parameters in Arachne mode, it may not be possible to slice these narrow areas. This is because the lines generated by Arachne also have a certain width range (to ensure the printer can physically extrude lines that wide). You can address this by reducing the min feature size and min wall width, but as you mentioned, this might degrade the print quality in other areas. You can also add a modifier to the teeth section to reduce the line width, allowing for smaller lines to be generated.

CraigHutchinson, 2024-11-14:
Thanks, I agree with the diagnosis. As I understand, as the feature is still >=25% to cause the tip of the teeth to print but the 2-layers below are somehow being discarded with a greater size which is odd. There is a mismatch on layer-by-layer slicing here which could also occur in other model geometries. 

It sort of smells like it could be occuring for other models but its hard to spot, having some sort of warning for floating regions of this sort could be a potential improvement. However, I'm happy enough if this is closed as won't fix for now.