bambulab/BambuStudio #5233: always scarf seam on inner walls

nicolasreynoso, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Ryzen 5 3600/RTX3060/32gb RAM

### Printer

Generic

### How to reproduce

1 enable scarf seam type : contorno
2 disable Scarf joint for inner wall.

### Actual results

scarf joint on all te walls. 

### Expected results

scarf joint on outer wall olny. 

### Project file & Debug log uploads

[Scarf.zip](https://github.com/user-attachments/files/17733682/Scarf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

nicolasreynoso, 2024-11-14:
[image]
[image]

QingZhangBambu, 2024-11-19:
@nicolasreynoso OK， i will check this

zkcreations, 2025-01-10:
I am having the same issue even with version 1.10.1.50. This problem has been here for a few months now when can we expect this to be fixed?

theTobster500, 2025-03-20:
still here on 1.10.2.76 two more months later. as shown in the screenshots, contour and hole doesn't have this issue

QingZhangBambu, 2025-03-21:
[image]
nope, i've fixed it on version 1.10.2

QingZhangBambu, 2025-03-21:
I'm sorry, I forgot to reply to this link

QingZhangBambu, 2025-04-02:
I will recheck this, sorry for the inconvenient

QingZhangBambu, 2025-04-02:
@theTobster500 and could you provide your model? that would be helpful
simple model works fine
[image]

theTobster500, 2025-04-06:
@QingZhangBambu before sending more info i wanted to check version 2.0 and I can confirm it is now working as it should

closed 2025-05-23 by QingZhangBambu