bambulab/BambuStudio #5237: Extreme Lag/Delay between "tabs"

dwissm1, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 15.1

### Additional system information

3 GHz 6-Core Intel Core i5
64 GB 2667 MHz DDR4

### Printer

P1S

### How to reproduce

1. Load STL file
2. Click Prepare
3. Click Preview ... very long delay to switch over. Rainbow Pinwheel of Death on macOS
4. Click device... very long delay to switch over. Rainbow Pinwheel of Death on macOS
5. Click back to Preview... same thing. Long delay, Rainbow Pinwheel of Death on macOS
6. Anytime I click between any tabs, the delay is 5 seconds or more.

### Actual results

Its just extremely slow and Rainbow Pinwheel of Death on macOS but if you wait it out, it will come back.

### Expected results

I upgraded today to the latest.  Didnt do this yesterday on the last version.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-14:
Hello, we are very sorry for the inconvenience caused to you. We are urgently investigating the issue~

HO44-PROJECT, 2024-11-15:
They just released 1.10.1.50 that closes my similar case.
https://github.com/bambulab/BambuStudio/issues/5194

dwissm1, 2024-12-09:
Seems to be fixed!  You guys are great.

closed 2025-01-10 by Haidiye00