bambulab/BambuStudio #5238: Import step file crashes bambu studio

vasichkin, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Additional system information

Macbook pro 2019

### Printer

Any

### How to reproduce

Import this step file

### Actual results

Bambu studio hangs, than crshes

### Expected results

File imported

### Project file & Debug log uploads

[step.zip](https://github.com/user-attachments/files/17737500/step.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DoTsTeR77, 2024-11-14:
yes, major issues with me importing step files that previously imported fine.. the beta had the same problem.. I had to revert back to previous version..

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu