bambulab/BambuStudio #5239: STP /STEP Inport error

Dade1984, 2024-11-13:
### Bambu Studio Version

10.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu x1c

### How to reproduce

STP /STEP Import error
software close

### Actual results

. STP /STEP Import error
software close

### Expected results

. STP /STEP Import error
software close

### Project file & Debug log uploads

STP /STEP Import error
software close

### Checklist of files to include

- [X] Log file
- [X] Project file

Stevelsch, 2024-11-14:
Related to https://github.com/bambulab/BambuStudio/issues/5261

DoTsTeR77, 2024-11-14:
yes, major issues with me importing step files that previously imported fine.. the beta had the same problem.. I had to revert back to previous version..

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

DoTsTeR77, 2024-11-15:
yup, just tried the new  package.. I still cant import some step files no matter how much I change those sliders.. I wish they would just put it back the way it was.. or put an option to import them the way the previous version did which gave me no issues..

DoTsTeR77, 2024-11-15:
> It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. [...]

yup, just tried the new package.. I still cant import some step files no matter how much I change those sliders.. I'd love to see it back the way it was.. or put an option to import them the way the previous version did which gave me no issues..

Unfortunately I cant provide the step files as it work under NDA..

ohh.. Could I ask what those sliders values would have been by default if the previous version used them?

Also.. all work on this is appreciated.. I dont want to sound like it isnt..

MackBambu, 2024-11-18:
> ohh.. Could I ask what those sliders values would have been by default if the previous version used them?

liner deflection is 0.03, angle deflection is 0.5.
All algorithms remain unchanged; in fact, only the parameter settings have been made configurable.

Hardrada, 2024-11-18:
This issue is making the app completely and utterly useless and it should be prioritized as an immediate fix. The software literally depends on step file imports and thus this should be the number one issue fixed. I've had to use orca slicer to get my prints done and then I can't import them back into bambu studio because of a file version mis-match.. This is killing me.

DoTsTeR77, 2024-11-18:
I second that issue in not ok and needs to be taken care of..   the step file import process needs to revert back to version 1.97.52 behavior.. it worked fine under that version..

DoTsTeR77, 2024-11-20:
> @Hardrada @DoTsTeR77 Deleting the configuration file will reset the parameters, but the logic for using the parameters remains unchanged from the previous version. I will wait for the feedback because I still don't know what happened.
> 
> > I second that issue in not ok and needs to be taken care of.. the step file import process needs to revert back to version 1.9,7.52 behavior.. it worked fine under that version..
> 
> I suspect you may have set an incorrect parameter that prevents successful meshing and enabled "Do not show again."

you suspect wrong.

MackBambu, 2024-11-20:
> not the case
@DoTsTeR77 
Would it be possible for you to record an operation video? This would help me reproduce the issue. Thank you.

nitrousdave, 2024-12-03:
I too have the same problem Hardrada does when trying to use a step file as a modifier. This was only after the file import dialog was added. Windows 10 and 11 computers both have the problem. Seems like the step file import dialog should pop up when using a step as a modifier.

Stevelsch, 2024-12-03:
Yes. The step import is ruined by this update. Still does not work. - and why do this change at all?Sent from my iPhone

MackBambu, 2025-04-29:
The new version has already switched to a different threading model. Does the issue still persist?

stvelsc, 2025-04-29:
It is supposed to look like this:

[image]

MackBambu, 2025-04-29:
> Just tested with the same part:
> [Shell_Part.txt](https://github.com/user-attachments/files/19952374/Shell_Part.txt)

Try reducing the linear deflection and angle deflection to achieve more subdivisions. It seems that 0.002 and 0.05 produce better results.
[image]

[image]

stvelsc, 2025-04-29:
Wow. Great thank you very much! I am back to using STEP then!