bambulab/BambuStudio #5240: Bambu Studio - Import STEP files

kvak1, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Additional system information

Cpu i5, memory 32 Gb

### Printer

P1S

### How to reproduce

If I try import any STEP file, the studio opens form "Step file important  parameters".  After choosing any action from the choices of the form the application waits - calculates and in the end finisches with information "Unhandled unknown exception; terminating the application. If I refuse correcting the file file and just press "Cancel" no file is imported. Other types of files like stl and 3mf work normaly.

### Actual results

Application works normaly but I have to use stl or 3mf files.

### Expected results

I need to work with STEP files.

### Project file & Debug log uploads

I hope no file is not needed..

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Trobinou, 2024-11-13:
I have the same bug on Windows 10, except that after clicking on the "OK" button in the error message, Bambu Studio closes automatically.

kvak1, 2024-11-13:
Yes, I have the same too, I forgot mention it. I realized another thing - if you switch of the choice "Show the step mesh parameter setting dialog" in General setting (ctrl-p), the application shows the message about its terminating directly without offering correcting your step file. It's sad but I am glad that is not my mistake. I hope it will be repaired soon. Thanks..

DoTsTeR77, 2024-11-14:
yes, major issues with me importing step files that previously imported fine.. the beta had the same problem.. I had to revert back to previous version..

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

kvak1, 2024-11-14:
> It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. [...]

Thank you very much Martin, I have just installed version (94) from your link. Steps work and temperature is visible (I didnt noticed that before). Yes I hope too that new release will be published soon but for this moment I am happy that I can use my steps. Karel

DoTsTeR77, 2024-11-15:
yup, just tried the new  package.. I still cant import some step files no matter how much I change those sliders.. I'd love to see it back the way it was.. or put an option to import them the way the previous version did which gave me no issues..

ohh.. Could I ask what those sliders values would have been by default if the previous version used them?

Also.. all work on this is appreciated.. I dont want to sound like it isnt..

DoTsTeR77, 2024-11-15:
> It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. [...]

yup, just tried the new package.. I still cant import some step files no matter how much I change those sliders.. I'd love to see it back the way it was.. or put an option to import them the way the previous version did which gave me no issues..

Unfortunately I cant provide the step files as it work under NDA..

ohh.. Could I ask what those sliders values would have been by default if the previous version used them?

Also.. all work on this is appreciated.. I dont want to sound like it isnt..

kvak1, 2024-11-15:
> yup, just tried the new package.. I still cant import some step files no matter how much I change those sliders.. I'd love to see it back the way it was.. or put an option to import them the way the [...]

I've just open the Studio and new version 1.10.1 is offered. Try it. There are more changes in it...

DoTsTeR77, 2024-11-19:
> I've just open the Studio and new version 1.10.1 is offered. Try it. There are more changes in it...

I actually did try 1.10.1 as well, same day I posted what I posted.. same behavior..