bambulab/BambuStudio #5241: Right Click Functionality Unavailable on MacBook M2 in Bambu Studio

skogsbergh, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

15.1 (24B83)

### Additional system information

Apple M2, 16GB of RAM

### Printer

Bambu Lab A1

### How to reproduce

Steps to Reproduce:

Open Bambu Studio version 1.10.0.89 on a MacBook M2 with macOS Sequoia 15.1.
Attempt to perform a right-click on any user interface element using the following methods:
Secondary click via Trackpad (two-finger tap)
Control + Click
Option + Click
Command + Click
Ensure that the "Secondary click" option is enabled in macOS System Settings under Trackpad preferences.
Observe that no context menu or secondary options appear upon attempting any of the right-click methods.

### Actual results

Actual Results: When attempting any right-click methods (two-finger tap, Control + Click, Option + Click, or Command + Click), there is no response from the application. The expected context menu or secondary options do not appear, and no visual or auditory feedback indicates that a right-click action has been recognized.

### Expected results

A context menu should appear, providing access to additional options relevant to the selected UI element.

### Project file & Debug log uploads

No specific log file was found for this issue. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-11-14:
[image]
I have test it with my M2 in V1.10.0.89, I can make sure two way works at least:
- Secondary click via Trackpad (two-finger tap)
- Control + Click

I want to remid you that both way's click should be the 'Real Click'（You must pressed down your trackpad）, at least I tried this way and succeed, but if I just 'touch' the trackpad it failed like your description.

Haidiye00, 2024-11-19:
> ![img_v3_02gk_716356ee-48e4-4a73-9092-cd538644e5dg](https://private-user-images.githubusercontent.com/19528955/386009563-038927ac-3571-4c32-a944-5b9f39b9ac3e.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5NzkzOTYsIm5iZiI6MTczMTk3OTA5NiwicGF0aCI6Ii8xOTUyODk1NS8zODYwMDk1NjMtMDM4OTI3YWMtMzU3MS00YzMyLWE5NDQtNWI5ZjM5YjlhYzNlLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDAxMTgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2MmMwN2YwNDUyYTBlMWIwYzVjYjliYTAwNjhjM2Y3YWFhNDhkNDJjYzYxNmRlYzQ5MTE2NTIxNjE4MjkyNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CBQFyF2uabc2m7WwTrz34DfR6KMy3R9YTJv1kA454wU) [ ![img_v3_02gk_716356ee-48e4-4a73-9092-cd538644e5dg](https://private-user-images.githubusercontent.com/19528955/386009563-038927ac-3571-4c32-a944-5b9f39b9ac3e.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5NzkzOTYsIm5iZiI6MTczMTk3OTA5NiwicGF0aCI6Ii8xOTUyODk1NS8zODYwMDk1NjMtMDM4OTI3YWMtMzU3MS00YzMyLWE5NDQtNWI5ZjM5YjlhYzNlLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDAxMTgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2MmMwN2YwNDUyYTBlMWIwYzVjYjliYTAwNjhjM2Y3YWFhNDhkNDJjYzYxNmRlYzQ5MTE2NTIxNjE4MjkyNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CBQFyF2uabc2m7WwTrz34DfR6KMy3R9YTJv1kA454wU) ](https://private-user-images.githubusercontent.com/19528955/386009563-038927ac-3571-4c32-a944-5b9f39b9ac3e.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5NzkzOTYsIm5iZiI6MTczMTk3OTA5NiwicGF0aCI6Ii8xOTUyODk1NS8zODYwMDk1NjMtMDM4OTI3YWMtMzU3MS00YzMyLWE5NDQtNWI5ZjM5YjlhYzNlLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDAxMTgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2MmMwN2YwNDUyYTBlMWIwYzVjYjliYTAwNjhjM2Y3YWFhNDhkNDJjYzYxNmRlYzQ5MTE2NTIxNjE4MjkyNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CBQFyF2uabc2m7WwTrz34DfR6KMy3R9YTJv1kA454wU) [ ](https://private-user-images.githubusercontent.com/19528955/386009563-038927ac-3571-4c32-a944-5b9f39b9ac3e.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5NzkzOTYsIm5iZiI6MTczMTk3OTA5NiwicGF0aCI6Ii8xOTUyODk1NS8zODYwMDk1NjMtMDM4OTI3YWMtMzU3MS00YzMyLWE5NDQtNWI5ZjM5YjlhYzNlLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDAxMTgxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2MmMwN2YwNDUyYTBlMWIwYzVjYjliYTAwNjhjM2Y3YWFhNDhkNDJjYzYxNmRlYzQ5MTE2NTIxNjE4MjkyNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CBQFyF2uabc2m7WwTrz34DfR6KMy3R9YTJv1kA454wU) I have test it with my M2 in V1.10.0.89, I can make sure two way works at least:
> 
> * Secondary click via Trackpad (two-finger tap)
> * Control + Click
> 
> I want to remid you that both way's click should be the 'Real Click'（You must pressed down your trackpad）, at least I tried this way and succeed, but if I just 'touch' the trackpad it failed like your description.

look this

skogsbergh, 2024-11-22:
I updated my MacBook to Sequoia 15.1.1 and the issue has been resolved. Thank you all for the responses.

closed 2024-11-22 by Haidiye00