bambulab/BambuStudio #5243: step file cant open

atr-ua, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 x64

### Printer

A1 mini

### How to reproduce

Ctrl+I
[image]
Ok not responded
Any slider change close window and follow next message box^
[image]

### Actual results

[image]
and application close

### Expected results

working with .step file

### Project file & Debug log uploads

[crash_Wed_Nov_13_22_00_53_0_done.log](https://github.com/user-attachments/files/17739283/crash_Wed_Nov_13_22_00_53_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

MartinChan3, 2024-11-14:
please see: https://github.com/bambulab/BambuStudio/issues/5186 , or switch language to English to solve this problem.

the0bone, 2024-11-14:
Same here. No Import possible

DoTsTeR77, 2024-11-14:
yes, major issues with me importing step files that previously imported fine.. the beta had the same problem.. I had to revert back to previous version..

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu