bambulab/BambuStudio #5244: 1.10.0 is Unuseable - Pinwheel of Death

jmthayer, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

13.7.1 (22H221)

### Printer

X1C

### How to reproduce

Install software,
Software becomes completely useless,
Pinwheel of death at every process or tab change

### Actual results

Unstable and unuseable

### Expected results

Needs to have the same or better performance than the previous releases

### Project file & Debug log uploads

[BambuLog.zip](https://github.com/user-attachments/files/17739355/BambuLog.zip)
Absolutely absurd that this is a mandatory field.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sxdjt, 2024-11-13:
+1

Seems to be most prevalent when working on a project with multiple plates.  Rolling back to the previous version until this is fixed.

jmthayer, 2024-11-13:
Rolled back to 1.9.5.51 and works perfectly on my machine.  This confirms it's an issue with the 1.10.0 version.  If others are having the same problem, recommend using the latest 1.9 version and skipping this release.

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

HO44-PROJECT, 2024-11-15:
They just released 1.10.1.50 that closes my similar case.
https://github.com/bambulab/BambuStudio/issues/5194

Haidiye00, 2024-11-20:
> They just released 1.10.1.50 that closes my similar case. #5194

please try this~

closed 2024-12-05 by Haidiye00