bambulab/BambuStudio #5245: Bambu Studio Device, AMS depiction doesn't show filament changes

tanery02048, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	Intel(R) Core(TM) i7-14700K   3.40 GHz
Installed RAM	32.0 GB (31.8 GB usable)
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Graphics          NVIDIA GeForce RTX4070 SUPER

### Printer

Bambu Labs P1S

### How to reproduce

Print any model where filament color is changed within the AMS.  

### Actual results

Upon changing, the new color is only shown on the page below the AMS grey box and the extruder until the filament is ready to change again, at which point the full path from the spool to the extruder is displayed.  Then the new color is once again only updated from the grey AMS box to the extruder.

### Expected results

In previous versions, the filament color is displayed on the Device AMS display from the Spool to the extruder upon each load.

### Project file & Debug log uploads

[Coaster_0.2nozzle.zip](https://github.com/user-attachments/files/17739422/Coaster_0.2nozzle.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SJChannel, 2024-11-15:
I am seeing this same problem on an Apple M1 running Sonoma 14.7.1 using Bambu Studio version 1.10.1.50.

walterwongbbl, 2024-11-21:
@tanery02048 
Can you provide pictures

SJChannel, 2024-11-21:
[image]

Here is one example.

SJChannel, 2024-11-23:
[image]

Here is a slightly different example.  In this case, it had been printing in red filament and then it switched to orange and then green.