bambulab/BambuStudio #5246: Tree Slim and Tree Hybrid May Generate Floating Support Structrues in 1.10.0.89

Kazeshima, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 23H2 Build 22631.4460

### Printer

Bambu A1

### How to reproduce

1. Select Bambu Lab A1 0.2 nozzole
2. Support: Enable Support. Type: Tree Auto. Style: Tree Slim.
3. Slice Plate

### Actual results

After upgraded to the latest version 1.10.0.89 the tree slim and tree hybrid option may generates broken supports, causing floating geometry. I have only observed this in the 0.2mm nozzle settings so far
[image]
[image]

### Expected results

The tree branches generated by the tree auto settings after slicing should have solid coherent connection in each layer.

### Project file & Debug log uploads

[shiroko_terror_bug_report.zip](https://github.com/user-attachments/files/17739690/shiroko_terror_bug_report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

VicisL, 2024-11-30:
Also having this issue on the P1S with a 0.2mm Nozzle profile.

Tynam, 2024-11-30:
Seconded; I'm having the same issue with an A1 and an 0.2mm hotend.  I was using my own settings so this isn't an issue in the presets; it's a problem with the slicer creating supports in 0.2mm.  Does not occur with larger nozzle sizes.
[image]

tangketan, 2024-12-09:
Thanks for reporting the issue. We are looking into it.

aotillar, 2024-12-28:
I am also currently having this issue.

ernestasKup, 2025-01-05:
Still an issue in 1.10.1.50 using the macBook.
Same repro steps when using the Slim and Organic tree supports (very important for the 0.2mm miniatures).

Could you please prioritise this bug? Or give us a workaround?
[image]

Tynam, 2025-01-05:
None of us said it explicitly but I should confirm that this is an issue in Bambu Studio 1.10 specifically.  At the moment I've reverted to 1.09 to avoid it.

(This is frustrating as I could really use the 1.10 features, but obviously they're no use if I can't use supports with 0.2mm detailed projects!)

I can provide some example .3mf files that demonstrate the problem well if that will help.

NothFie, 2025-01-06:
@Tynam @ernestasKup I'm sorry for the bad experience you had. We have already found out the the cause of the problem and I promise it will be fixed in the next version of Bambu Studio.

closed 2025-01-06 by NothFie