bambulab/BambuStudio #5248: BambuStudio crashes on MAC MINI M4 Pro when loading STEP file

Godsplan, 2024-11-13:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 15.1 (24B2082)

### Additional system information

M4 Pro 24 GB

### Printer

X1C

### How to reproduce

1. Goto "Prepare"
2. Add model
3. STEP file import parameters open
4. When trying to press "OK" 
[image]

### Actual results

Bambu Studio Crashes

### Expected results

It should load the STEP file

### Project file & Debug log uploads

{"app_name":"BambuStudio","timestamp":"2024-11-13 23:03:32.00 +0100","app_version":"01.10.00.89","slice_uuid":"891a1287-d359-3534-ba03-1c3efa3bbb66","build_version":"","platform":1,"bundleID":"com.bambulab.bambu-studio","share_with_app_devs":0,"is_first_party":0,"bug_type":"309","os_version":"macOS 15.1 (24B2082)","roots_installed":0,"name":"BambuStudio","incident_id":"6ACE6F61-B6CA-457A-9FE0-F28171D21E35"}
{
  "uptime" : 77000,
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "Mac16,11",
  "coalitionID" : 12367,
  "osVersion" : {
    "train" : "macOS 15.1",
    "build" : "24B2082",
    "releaseType" : "User"
  },
  "captureTime" : "2024-11-13 23:03:32.3492 +0100",
  "codeSigningMonitor" : 2,
  "incident" : "6ACE6F61-B6CA-457A-9FE0-F28171D21E35",
  "pid" : 54580,
  "translated" : false,
  "cpuType" : "ARM-64",
  "roots_installed" : 0,
  "bug_type" : "309",
  "procLaunch" : "2024-11-13 23:03:21.5258 +0100",
  "procStartAbsTime" : 1849142190185,
  "procExitAbsTime" : 1849401138733,
  "procName" : "BambuStudio",
  "procPath" : "\/Applications\/BambuStudio.app\/Contents\/MacOS\/BambuStudio",
  "bundleInfo" : {"CFBundleShortVersionString":"01.10.00.89","CFBundleVersion":"","CFBundleIdentifier":"com.bambulab.bambu-studio"},
  "storeInfo" : {"deviceIdentifierForVendor":"1ABF3BAA-E3E5-5692-9C31-434984E4AAEE","thirdParty":true},
  "parentProc" : "launchd",
  "parentPid" : 1,
  "coalitionName" : "com.bambulab.bambu-studio",
[... 264 lines cut ...]
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB7206 Smart Hub
USB Device: Logitech StreamCam
USB Device: USB5734
USB Device: USB2.0 Hub
USB Device: USB4206 Smart Hub
USB Device: Wireless Receiver
USB Device: HP Thunderbolt Dock G4
USB Device: USB2734
USB Device: USB2.0 Device
USB Device: USB31Bus
USB Device: USB31Bus
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Sep 23 2024 19:01:27 version 23.900.1.1.41.51.169 FWID 01-09ad7e95
IO80211_driverkit-1315.7 "IO80211_driverkit-1315.7" Sep 25 2024 21:54:21
AirPort: 
Network Service: Wi-Fi, AirPort, en1
PCI Card: ethernet, Ethernet Controller, Thunderbolt@3,0,0
Thunderbolt Device: Thunderbolt Dock G4, HP, 1, 38.1

### Checklist of files to include

- [X] Log file
- [ ] Project file

MartinChan3, 2024-11-14:
please see: https://github.com/bambulab/BambuStudio/issues/5186 or switch language to English to solve this question befoer Studio fix this.

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50