bambulab/BambuStudio #5252: Image built with DockerBuild.sh not working : "libavcodec.so.61" not found

sylvaintremblay666, 2024-11-14:
### Bambu Studio Version

v01.10.00.89

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 23.10

### Printer

-

### How to reproduce

- clone github repo
- checkout tag: v01.10.00.89
- cd into BambuStudio and run: ./DockerBuild.sh
- start the container: ./DockerRun.sh

### Actual results

the following error:

/BambuStudio/build/package/bin/bambu-studio: error while loading shared libraries: libavcodec.so.61: cannot open shared object file: No such file or directory

### Expected results

bambuStudio should start.

**Workaround**

The `libavcodec.so.61` lib is built during the dependencies build phase and can be found in `/BambuStudio/build/package/bin`. Add an `LD_LIBRARY_PATH` env var pointing to this location in the `DockerRun.sh` script then BambuStudio will start : 
```
...
-e LD_LIBRARY_PATH=/BambuStudio/build/package/bin \
...
```

### Project file & Debug log uploads

-

### Checklist of files to include

- [X] Log file
- [ ] Project file

saberworks, 2024-11-15:
I'm using firejail to sandbox this app and I'm getting the same error.  I used appimage-extract to extract the appimage and then run it like this:
```
brian@link:~/Desktop/squashfs-root$ firejail ./bin/bambu-studio 
Reading profile /etc/firejail/default.profile
Reading profile /etc/firejail/disable-common.inc
Reading profile /etc/firejail/disable-passwdmgr.inc
Reading profile /etc/firejail/disable-programs.inc
Warning: networking feature is disabled in Firejail configuration file

** Note: you can use --noprofile to disable default.profile **

Parent pid 6027, child pid 6028
Warning: not remounting /var/lib/docker/overlay2/e43d071a928bd1ebb2669851245eb4df8256a81ef2ad4a3d42b2c62a1191efca/merged
Warning: not remounting /var/lib/docker/overlay2/f4147a68b0d96ba3bab702b9742a507a8b8f07c2e069997e6ad6486ae28162fa/merged
Warning: not remounting /var/lib/docker/overlay2/e43d071a928bd1ebb2669851245eb4df8256a81ef2ad4a3d42b2c62a1191efca/merged
Warning: not remounting /var/lib/docker/overlay2/f4147a68b0d96ba3bab702b9742a507a8b8f07c2e069997e6ad6486ae28162fa/merged
Warning: cleaning all supplementary groups
Child process initialized in 36.92 ms
./bin/bambu-studio: error while loading shared libraries: libavcodec.so.61: cannot open shared object file: No such file or directory

Parent is shutting down, bye...
```
The older version of this appimage worked fine like this.  The workaround posted here doesn't quite help because there's no DockerRun.sh in the appimage.

sylvaintremblay666, 2024-11-16:
> I'm using firejail to sandbox this app and I'm getting the same error. I used appimage-extract to extract the appimage and then run it like this:
>
> ```
> brian@link:~/Desktop/squashfs-root$ firejail [...]

You can probably work around this issue in a similar way... (not 100% sure as I never used firejail but its worth a try)  :

Look for `libavcodec.so.61` in your extracted filetree, then launch bambu studio like this:
```
$ LD_LIBRARY_PATH=./<folder where is libavcodec.so.6> firejail ./bin/bambu-studio
```

MackBambu, 2024-11-24:
Due to the addition of FFmpeg, the Dockerfile needs to be fixed.

gw0, 2024-12-30:
I ran into the same issue using `DockerRun.sh` with Bambu Studio latest `master` (commit `c1b3cc` 01.10.1-next). The solution by setting `LD_LIBRARY_PATH` env var works.

MackBambu, 2025-03-31:
https://github.com/bambulab/BambuStudio/pull/5156
This patch fixes all the issues. I think it can be closed now.

closed 2025-03-31 by MackBambu