bambulab/BambuStudio #5253: Pause at layer not working

donhohler, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

I add a pause at any layer. Then slice the file. Examine the g-code. There is no pause. This just started happening with the latest update.

### Actual results

When I print the file there is no pause.

### Expected results

There should have been a pause.

### Project file & Debug log uploads

[Intake Bracket.zip](https://github.com/user-attachments/files/17742291/Intake.Bracket.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

donhohler, 2024-11-14:
I also tried adding custom g-code. M400 U1. Nothing. It did not get added to the g-code.

donhohler, 2024-11-14:
I reverted back to 1.9.7.52 and everything works.

MartinChan3, 2024-11-14:
[image]
It is the same, it seems that all things are right with my Studio and X1C...

donhohler, 2024-11-14:
There is definitely a problem. I reinstalled 1.10 and got the same problem again.

MartinChan3, 2024-11-15:
I believe Bambu will fix this issue in next version.

donhohler, 2024-11-18:
Yes. I could not connect to my printer with it. No matter what I tried. I'm back to 1.9 again.

DavidLorentzen, 2024-11-19:
I have 1.10.1 and experienced the same problem, just an hour ago. The first print paused correctly, and I imported another object of the same specs, and pause was skipped. The pause was supposed to be at the same layer (19) as the first. Both prints were imported as 3mf files.

XunZhangBambu, 2024-11-19:
@DavidLorentzen Could you please provide the 3mf?

donhohler, 2024-11-19:
The pause problem was supposedly fixed in the latest version. However, when I tried it, it would not connect to the printer. So, I can not confirm whether it's fixed or not. I am currently still using ver 1.9. It would be nice if Bambu would do a little in house testing before throwing these updates out in the wild for all of us to test for them.

DavidLorentzen, 2024-11-19:
[Aaron-3mf.3mf.txt](https://github.com/user-attachments/files/17808200/Aaron-3mf.3mf.txt)

I added a .txt to the end of the file name to upload.

Dave

DavidLorentzen, 2024-11-24:
Encountered the problem again yesterday, different file. I set a pause in order to insert a nut. It blew right past the pause and ran to the end of the print. STL is attached.

[Pivot+Handle+-+SAE+-+Quarter20.stl.txt](https://github.com/user-attachments/files/17894304/Pivot%2BHandle%2B-%2BSAE%2B-%2BQuarter20.stl.txt)

XunZhangBambu, 2024-11-25:
@DavidLorentzen  
Hello, I tried the file you provided and added the pause G-code, but I couldn't reproduce your issue. You should be able to directly provide the 3MF file with parameters without needing to change the file extension.

[image]
[image]

DavidLorentzen, 2024-11-25:
Sorry, I've been working with STL files, and didn't save all my settings as 3MF. I've tried unsuccessfully to recreate the problem after two attempts.