bambulab/BambuStudio #5254: Models small part is overheated

siavpere, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1

### Printer

Bambu LAb A1

### How to reproduce

1.- Load the project
2.- Send to print (it's a quick print 12 mins)

[A1_TAPACUBOS_TRAS.3mf.zip](https://github.com/user-attachments/files/17742511/A1_TAPACUBOS_TRAS.3mf.zip)

### Actual results

In general the piece is good, but the center hub is overheated.
I tried to slow down in this part but 'Small perimeter' and 'Small perimetr threshold' is not working.

[FAILED_PRINT.HEIC.zip](https://github.com/user-attachments/files/17742519/FAILED_PRINT.HEIC.zip)

### Expected results

[Expected_result.HEIC.zip](https://github.com/user-attachments/files/17742522/Expected_result.HEIC.zip)

### Project file & Debug log uploads

[A1_TAPACUBOS_TRAS.3mf.zip](https://github.com/user-attachments/files/17742539/A1_TAPACUBOS_TRAS.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

siavpere, 2024-11-14:
I send to you an screenshot of the principal speed. When the speed decreases the center hub starts to fail.

[Captura de pantalla 2024-11-13 a la(s) 9.54.44 p.m..zip](https://github.com/user-attachments/files/17742611/Captura.de.pantalla.2024-11-13.a.la.s.9.54.44.p.m.zip)

XunZhangBambu, 2024-11-25:
@siavpere  The speed you set for “small perimeter” may be adjusted by some subsequent features. You can disable these features to ensure that the speed you set takes effect. For example ,"smooth speed discontinuity" in process tab and "Slow printing down for better layer cooling" in filmament cooling tab.I disabled those settings, and the speed I set took effect successfully.

[image]