bambulab/BambuStudio #5255: mac端容易卡死

zeng89, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.6.1

### Printer

X1Carbon

### How to reproduce

下载好复杂模型后点击切片

### Actual results

然后就卡死，导致程序无响应

### Expected results

可以正常打印

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-14:
Hello, I'm very sorry for the inconvenience caused to you. Excuse me, is it a multi plate project? We found that single plate projects do not work.

pridi, 2024-11-14:
> Hello, I'm very sorry for the inconvenience caused to you. Excuse me, is it a multi plate project? We found that single plate projects do not work.

Hello, yes, it is a multi plate project. I can confirm that by **using only one plate** I do not have the problem. Hope the problem with multi plate project will be fixed soon!

zeng89, 2024-11-14:
> I have the same issue!

1.10.0.89 will be ok，you can try this

zeng89, 2024-11-14:
> Hello, I'm very sorry for the inconvenience caused to you. Excuse me, is it a multi plate project? We found that single plate projects do not work.

you can try version  1.10.0.89

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

closed 2024-11-15 by zeng89

pridi, 2024-11-15:
> It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. [...]

Thank you very much! It's working perfectly