bambulab/BambuStudio #5256: Weird lines in slice for project

armaans96, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.4.1

### Additional system information

1.4 GHz Quad-Core Intel Core i5
Intel Iris Plus Graphics 645 1536 MB
8 GB 2133 MHz LPDDR3

### Printer

a1 mini

### How to reproduce

slice produces blue lines which are prints across the infill which don't seem to have any purpose.
[image]

### Actual results

printer printed the lines as expected

### Expected results

N/A

### Project file & Debug log uploads

[EDC Holder V3.3mf.zip](https://github.com/user-attachments/files/17745378/EDC.Holder.V3.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

CraigHutchinson, 2024-11-14:
The blue lines are 'bridge' which is necessary to suport the 'internal solid infill' on the layer above. The bridge lines extend to the 'sparse' infill as a support points for the later top layers. Without this layer you would be printing mid-air. TLDR; the bridge layer is necessary.

QingZhangBambu, 2024-11-15:
@XunZhangBambu please check this