bambulab/BambuStudio #5258: Import STEP chrashes when using Swedish language

lillknurra, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Seqoia

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Set language to Swedish
2. Start new project
3. Import STEP file

### Actual results

Application crashes

### Expected results

No crash

### Project file & Debug log uploads

Not possible since the model file can't be imported...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

lillknurra, 2024-11-14:
I'm running with the M2 chip though...

MartinChan3, 2024-11-14:
Believe me it works!

closed 2024-11-18 by MackBambu