bambulab/BambuStudio #5259: Edit text of a text modifier messes up orientation

elsni, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1, Bambu Lab P1S

### How to reproduce

1. Load a .3mf model with a text modifier from previous version
2. select text modifier
3. change text

### Actual results

Text is messed up, sometimes completely out of position, sometimes weird orientation.

### Expected results

Text changes but keeps orientation and position.

### Project file & Debug log uploads

[BambuBug_Text.zip](https://github.com/user-attachments/files/17747892/BambuBug_Text.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-11-15:
We are optimizing the function of maintaining the current pose of this text. Sorry for the inconvenience caused to you

Haidiye00, 2024-11-20:
> Also deselecting uniform scaling does not deselect it in version 1.10.1.50. boundary box is in wrong orientation. totally ununsable.

Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you