bambulab/BambuStudio #5260: AMS Status display delayed

Udo-Nbg, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Load Filament via AMS and watch the Filamentpath in Studio
You Can See the Filament arrived in the Hotend but not (drawn) the Path from the spool over the HUB to the hotend is displayed.
After the printing has started the Filamentpath will correct updated.

In the Studio Version bevore it was okay.

BTW: the horizontal line for the external Filament ist not straight, it has a small angle (1 pixel)

### Actual results

[image]

### Expected results

Display the Filamentpath just in time

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

an565, 2024-11-16:
[image]
Sometimes it gets mixed up.

walterwongbbl, 2024-11-21:
@Udo-Nbg 
At certain resolutions, UI resolution calculations may lose some accuracy, and we will address this issue.