bambulab/BambuStudio #5262: Split objects deletes parts

Thimosz, 2024-11-14:
### Bambu Studio Version

01.10.00.89

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

A1

### How to reproduce

1.Load a model with multiple objects
2. Split to objects or Parts
3. Small parts are deleted

### Actual results

Parts are deleted from the model.

### Expected results

Parts stay

### Project file & Debug log uploads

[BugReport.zip](https://github.com/user-attachments/files/17750215/BugReport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-11-15:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
And can you provide your 3mf?

Thimosz, 2024-11-15:
[Bug Splitting Objects.zip](https://github.com/user-attachments/files/17772121/Bug.Splitting.Objects.zip)
[image]

Thimosz, 2024-11-15:
I can do the same on my other system wich is running the previous version. With the exact same model and scaling there is not the issue.

Haidiye00, 2024-11-15:
Hello, there is indeed an issue. We apologize for any inconvenience caused and will fix it as soon as possible. At present, you can find an old version such as version 1.8, split it and import the new version for editing.

Haidiye00, 2024-11-15:
[image]
I found that this model has been scaled by you, resulting in small volume filtering. You can export and import the current 3mf model, and then split it, and there will be no problem. Of course, we will also consider your case
[image]

Thimosz, 2024-11-15:
Thanks for the work around. Yes i did scaled it as i designed it myself. Still strange in my opinion that its no problem in the older versions.
Can't wait for you guys to fix it :)

Haidiye00, 2024-11-15:
We will do it as soon as possible, thank you

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00