bambulab/BambuStudio #5264: Hotfix Preview may help those suffered from Mac (Intel CPU), STEP import crash and loss of temp on Device tab...

MartinChan3, 2024-11-14:
### Bambu Studio Version

V1.10.0 

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS

### Printer

Bambu Studio A1

### How to reproduce

It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

### Actual results

I hope everything will be all right.

### Expected results

May the force be with you!

### Project file & Debug log uploads

None

### Checklist of files to include

- [X] Log file
- [X] Project file

FallasAutomation, 2024-11-14:
This fixed my temp display issues, Thanks!

JohnSmithCYM, 2024-11-18:
@ttt888zzz  Now this issue can be closed.

MartinChan3, 2024-11-20:
See more in V1.10.1 Public Release(Hotfix)

closed 2024-11-20 by MartinChan3