bambulab/BambuStudio #5266: Bug with STEP files

easysofts, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 24h

### Printer

X1C

### How to reproduce

Bug on importing STEP file,
When I click Ok, bambu studio crash
when i test to change setting on importing, bambu studio crash

### Actual results

When I click Ok, bambu studio crash
when i test to change setting on importing, bambu studio crash

### Expected results

Importing without bug like previous version

### Project file & Debug log uploads

2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.394797[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.561622[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.561622[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
2024-11-14 17:25:32.779244[Thread 0x00002968]:initialization for CFF / Type2 fonts (OTF)
[... 265 lines cut ...]
2024-11-14 17:25:36.899548[Thread 0x00005c4c]:command_get_access_code
2024-11-14 17:25:36.900131[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::load_ams_list: 1 items
2024-11-14 17:25:36.909169[Thread 0x00005c4c]:command_get_version
2024-11-14 17:25:36.910138[Thread 0x00005c4c]:command_get_access_code
2024-11-14 17:25:36.911136[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::load_ams_list: 1 items
2024-11-14 17:25:37.914307[Thread 0x00005c4c]:Slic3r::MachineObject::update_printer_preset_name 5322start update preset_name
2024-11-14 17:25:37.914307[Thread 0x00005c4c]:Slic3r::PresetBundle::get_printer_names_by_printer_type_and_nozzle 1855 printer name: Bambu Lab X1 Carbon 0.4 nozzle
2024-11-14 17:25:37.915306[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 0 id GFL99 color F98C36FF
2024-11-14 17:25:37.915306[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 1 id GFL99 color FFFFFFFF
2024-11-14 17:25:37.915306[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 2 id GFA00 color 00AE42FF
2024-11-14 17:25:37.915306[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 3 id GFL99 color 161616FF
2024-11-14 17:25:37.915306[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::load_ams_list: 5 items
2024-11-14 17:25:37.930893[Thread 0x00005c4c]:command_get_version
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:command_get_version
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:command_get_access_code
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 0 id GFL99 color F98C36FF
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 1 id GFL99 color FFFFFFFF
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 2 id GFA00 color 00AE42FF
2024-11-14 17:25:38.079209[Thread 0x00005c4c]:Slic3r::GUI::Sidebar::build_filament_ams_list: ams 0 tray 3 id GFL99 color 161616FF
2024-11

### Checklist of files to include

- [X] Log file
- [ ] Project file

MartinChan3, 2024-11-14:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu