bambulab/BambuStudio #5267: "Failed to connect to the server" after waking from sleep

samuelzamvil, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1

### Additional system information

16" MacBook Pro - M1 Pro - 32GB RAM

### Printer

Bambu Lab X1C

### How to reproduce

1. Use Bambu Studio
2. Put laptop to sleep
3. Upon coming back the Device pane doesn't work

### Actual results

Device pane shows "Failed to connect to the server". Restarting the app fixes the issue but requires saving and reloading projects.

### Expected results

Server connection is automatically reattempted/restored. Another option would be to add a "reattempt to connect" button for manually reattempting the connection. A timer button lock feature can be added to prevent overloading downstream services.

### Project file & Debug log uploads

No folder $HOME/Library/Application Support/BambuStudio/log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

minsis, 2025-03-15:
This issue is a ridiculous one to have. Just put some kind of refresh button in the app that allows us to refresh the connection without having to restart the app every couple of hours.

th3tick, 2025-04-26:
After the update where I saw what I thought was something indicating that this had been fixed, it actually seems to be a lot worse with about half my windows unable to connect "to the server" after I re-open my laptop.

Feels like I'm running Windows from the 1990s "have you turned it off and back on again?".