bambulab/BambuStudio #5268: Bambu-Studio

Hoelenmensch, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 / 11

### Additional system information

AMD Ryzen 7 5700U with Radeon Graphics            1.80 GHz
16,0 GB (15,4 GB verwendbar)
64-Bit-Betriebssystem, x64-basierter Prozessor
Windows 11 Home

### Printer

X1C / P1s

### How to reproduce

1. Datei importieren
2. ausgewählte step - datei import
3.  Parameters setzen
4. okay drücken

[image]

### Actual results

Das bambu - Studio bricht ab
[image]

### Expected results

es sollte die ausgewählte step - datei geöfnnet werden.
was nicht möglich ist
[image]

### Project file & Debug log uploads

[connection.zip](https://github.com/user-attachments/files/17754442/connection.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Hoelenmensch, 2024-11-14:
in der version 1.09.0752
hat alles geklapt

MartinChan3, 2024-11-15:
It seems that Studio has released a temporary package to provide an interim solution for those experiencing issues on Mac (Intel chip), STEP import crashes, and missing temperature displays. Hopefully, they will release the actual hotfix soon!
Windows：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_win_public-v01.10.00.94-20241114223322.exe
Mac：https://github.com/bambulab/BambuStudio/releases/download/v01.10.00.89/Bambu_Studio_mac-v01.10.00.94-20241114223322.dmg

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu