bambulab/BambuStudio #5272: Error when importing STEP-files

Runar80, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Enhetsnavn	LAPTOP-EOB499PU
Prosessor	AMD Ryzen 3 3200U with Radeon Vega Mobile Gfx     2.60 GHz
Installert RAM	12,0 GB (9,94 GB brukbar)
Enhets-ID	A6F98ACC-D6F7-41BD-B82F-C3F19F67F812
Produkt-ID	00356-02086-58594-AAOEM
Systemtype	64-biters operativsystem, x64-basert prosessor

### Printer

P1S

### How to reproduce

When importing STEP files the new dalog box asks you to define the refinement sliders. Dragging one slider makes the software calculate for an infinate amount of time. Touching the other slider instantly crashes the software. Error prompt: Unhandled unknown exception; terminating the aplication. This happends regardless of file size or refinement chosen

### Actual results

Bambu studio closes

### Expected results

Should import file to prepare area of bambu studio

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17756827/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Almisuifre, 2024-11-14:
I have exactly the same issue. Either it crashes or it doesn't open. It's unusable in its current state!

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu