bambulab/BambuStudio #5273: Seamissue

POCONOPRINTING, 2024-11-14:
### Bambu Studio Version

1.10

### Where is the application from?

Bambu Lab github releases

### OS version

Windows

### Additional system information

intel i9

### Printer

p1s

### How to reproduce

Seam alignment is not working correctly. Even if I paint the seam isn't not aligning
[image]
[image]
 properly. 

### Actual results

seam is the same no matter what setting I chose. 

### Expected results

The seam should be in a nice straight line. Its only in the most recent version. 

### Project file & Debug log uploads

[dishwasher_tab_storage_lid.zip](https://github.com/user-attachments/files/17756852/dishwasher_tab_storage_lid.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-11-15:
It is due to 1.10.0 start to use Scarf Seam by default, the Scarf Seam rendering was not like common seam.You can close seam option under filament setting.
[image]

QingZhangBambu, 2024-12-02:
@POCONOPRINTING render issue, the latest version had closed the scarf seam

closed 2025-04-03 by QingZhangBambu