bambulab/BambuStudio #5275: Pause commands not being honored

nate3D, 2024-11-14:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon 0.4 nozzle

### How to reproduce

Slice the plate, and right-click the layer slider to add a Pause. Print plate.

### Actual results

The printer ignores the pause command and simply finished the print as normal.

### Expected results

Pause at the specified layer and wait for a continue to be issued.

### Project file & Debug log uploads

[Fidget Wheels Bambu.zip](https://github.com/user-attachments/files/17757800/Fidget.Wheels.Bambu.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

donhohler, 2024-11-14:
Reported this yesterday. I don't think they believe me.
https://github.com/bambulab/BambuStudio/issues/5253

nate3D, 2024-11-14:
I tried the EXACT same profile just again and it did pause @donhohler . I have no idea I can't even reproduce it but I was standing at the printer waiting for the pause the first time and I can confirm it never happened. Sorry Bambu team - I'm an engineer and I know this isn't very helpful since we can't repro.

donhohler, 2024-11-14:
1.10 has not paused for me. I have tried 3 times. I am printing PPA/CF so I am getting tired of wasting it. This fails my print each time since I am inserting a nut in the print. Hence the need for the pause.

donhohler, 2024-11-14:
I now examine the g-code. If there is no M400 U1, then I don't bother printing. I have reverted back to 1.9.7.52.

XunZhangBambu, 2024-11-15:
@donhohler  We have identified the cause of the issue. Could you provide the original information of the 3mf file (e.g., the version number when saved by bambu studio), as we have noticed that the height at the insertion point is 12.20, while the saved height in the file is 12.04.

donhohler, 2024-11-15:
The file was originally created with ver. 1.9.7.52. I then upgraded to 1.10. Reopened the file and the problem appeared.

spuder, 2025-08-09:
I'm seeing the same issue on bambu studio `2.1.1.52`

[TPU Swap.gcode.3mf.zip](https://github.com/user-attachments/files/21700024/TPU.Swap.gcode.3mf.zip)

I'm using the stock change filament g-code except I've added an extra `M400 U1` on line 23

[image]

<details><summary>code</summary>
<p>

```
M620 S[next_extruder]A
M204 S9000
G1 Z{max_layer_z + 3.0} F1200

G1 X70 F21000
G1 Y245
G1 Y265 F3000
M400
M106 P1 S0
M106 P2 S0
{if old_filament_temp > 142 && next_extruder < 255}
    M104 S[old_filament_temp]
{endif}
{if long_retractions_when_cut[previous_extruder]}
    M620.11 S1 I[previous_extruder] E-{retraction_distances_when_cut[previous_extruder]} F{old_filament_e_feedrate}
{else}
    M620.11 S0
{endif}
M400

; ADD PAUSE AFTER CUTTING FOR EXTERNAL SPOOL
; This will pause the printer after cutting filament so user can manually handle external spool
M400 U1

G1 X90 F3000
G1 Y255 F4000
G1 X100 F5000
G1 X120 F15000
G1 X20 Y50 F21000
G1 Y-3
{if toolchange_count == 2}
    ; get travel path for change filament
    M620.1 X[travel_point_1_x] Y[travel_point_1_y] F21000 P0
    M620.1 X[travel_point_2_x] Y[travel_point_2_y] F21000 P1
    M620.1 X[travel_point_3_x] Y[travel_point_3_y] F21000 P2
{endif}
M620.1 E F[old_filament_e_feedrate] T{nozzle_temperature_range_high[previous_extruder]}
T[next_extruder]
M620.1 E F[new_filament_e_feedrate] T{nozzle_temperature_range_high[next_extruder]}

{if next_extruder < 255}
    {if long_retractions_when_cut[previous_extruder]}
        M620.11 S1 I[previous_extruder] E{retraction_distances_when_cut[previous_extruder]} F{old_filament_e_feedrate}
        M628 S1
        G92 E0
        G1 E{retraction_distances_when_cut[previous_extruder]} F[old_filament_e_feedrate]
        M400
        M629 S1
    {else}
        M620.11 S0
    {endif}
    
    ; PAUSE AFTER FILAMENT RETRACTION FOR EXTERNAL SPOOL
    ; Allow user to manually handle the external spool after retraction
    M400 U1
    G92 E0
    {if flush_length_1 > 1}
        M83
[... 121 lines cut ...]
    M400
    G1 Z{max_layer_z + 3.0} F3000
    {if layer_z <= (initial_layer_print_height + 0.001)}
        M204 S[initial_layer_acceleration]
    {else}
        M204 S[default_acceleration]
    {endif}
{else}
    ; PAUSE FOR SINGLE EXTRUDER FILAMENT CHANGE (P1S)
    ; Allow user to manually change filament on external spool
    M400 U1
    G1 X[x_after_toolchange] Y[y_after_toolchange] Z[z_after_toolchange] F12000
{endif}
M621 S[next_extruder]A

```
</details>

The rendered g-code has the pause on line 1081

[image]

Despite the `M400 U1` command being present, the pause is ignored on the printer. 

Printer Firmware: 1.7.0.0

AndiVo, 2025-09-01:
📝 Bug description

I can confirm this issue on a Bambu Lab P1S running with Bambu Studio 2.2.1.60.
When inserting a pause at layer via Studio (or when manually adding pause commands such as M400 / M400 U1 / M25), the printer completely ignores the pause and continues printing until the very last layer.

This happens regardless of whether the pause command is placed at the beginning of a layer or in mid-print. The expectation would be that the printer moves the toolhead away and waits for user interaction, but instead it simply finishes the job.

✅ Expected behavior

Printer should stop at the specified layer/command.

Toolhead should park, keep heaters active, and wait for user confirmation (resume).

❌ Actual behavior

Printer ignores the pause command.

Job continues until completion with no user intervention possible.

🔧 Workaround

Export G-code to SD card and start print from SD directly.
In SD-card mode, standard G-codes like M25 or M400 U1 are interpreted correctly and the pause works.

If the print is started via Bambu Studio / LAN / Cloud, pause commands are filtered or ignored.

📌 Notes

This makes planned filament changes (without AMS) or inserting magnets/nuts in prints unreliable.

The bug has been reproduced consistently on my setup.

It looks like the job wrapper generated by Studio bypasses or removes pause instructions.