bambulab/BambuStudio #5278: The preview images of the files saved on the SD card often fail to load.

836043550, 2024-11-15:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab github releases

### OS version

Windows10

### Additional system information

It doesn't matter.

### Printer

bambu lab A1

### How to reproduce

[image]

### Actual results

The preview images of the files saved on the SD card often fail to load.Why should problem reports be submitted on GitHub? The vast majority of users in China are unable to log in to this website, and countless machines cannot provide feedback on problems through this channel.

### Expected results

bingo~！

### Project file & Debug log uploads

[debug_Wed_Nov_13_14_30_28_13284.log.zip](https://github.com/user-attachments/files/17759920/debug_Wed_Nov_13_14_30_28_13284.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

gaolegao-lx, 2024-11-15:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

fahaaaxiki, 2024-11-27:
Hi, According to the picture you provided, the duration and consumable weight of all the projects that did not show the preview are 0, which may be due to the poor network of the printer causing the model file loading failure. You can try to check the strength of your network or restart the printer to solve the problem
[image]