bambulab/BambuStudio #5280: Show the actual (approximate) local Time a print job will finish

SirMest, 2024-11-15:
In BambuStudio, in the Device window, in the Printing Progress dashboard. Is it possible to show the approximate (local) time the print job will finish so we don't have to work out the remaining time + the current time?
Also, this is a minor bug in the very latest update (1.10.0.89) that makes the top of the Layer and remaining time text in the Print Progress panel cropped.

[image]

SirMest, 2024-11-15:
Apologies I meant to say the finish times on the same line as the remaining time

AbandonedRickshaw, 2024-11-15:
Why is this tagged as a Feature and not a bug?

teodoryantcheff, 2024-11-16:
This may be related. 
Layer indication is intact, no finish time, though.
And, yes -- I also do think this is a bug, not a feature.
 
[image]