bambulab/BambuStudio #5282: error  opening step file

Zedral1, 2024-11-15:
### Bambu Studio Version

V1.10.0.89

### Where is the application from?

Bambu Lab github releases, Built from source by yourself or others

### OS version

WIN 10

### Additional system information

Procesor	AMD Ryzen 7 5700G with Radeon Graphics            4.30 GHz
Nainstalovaná paměť RAM	64,0 GB (použitelné: 63,8 GB)
Typ systému	64bitový operační systém, procesor pro platformu x64
Edice	Windows 10 Pro
Verze	22H2
Datum instalace	‎27.‎12.‎2021
Build operačního systému	19045.5131
Výkon	Windows Feature Experience Pack 1000.19060.1000.0

### Printer

Bambu X1C

### How to reproduce

1 ,go to open new 
2 select step file 
- automaticly opened window "Step file import parameters"  
3 select OK 
- bam

### Actual results

[image]
[image]

### Expected results

after bambu studio shutig down
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17770808/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

threadstone83, 2024-11-15:
https://github.com/bambulab/BambuStudio/issues/5239#issuecomment-2476807705

MackBambu, 2024-11-18:
fixed
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-18 by MackBambu