bambulab/BambuStudio #5283: Auto Calibration for A1 Not Working

Dalydoo, 2024-11-15:
### Bambu Studio Version

1.10.0.89 

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

1) No "AutoCalibration" for A1 on the "Calibration" page, still manual calibration only. 
2) No "Finish Time" when printing on the "Device" page.

### Actual results

1) No Auto option
2) Print item and no Finish Time (Blank)

### Expected results

1) Auto Calibration
2) Print Finish Time visable

### Project file & Debug log uploads

No files are needed. Many are complaining about this on YouTube as the new software update promotes the addition of these new features in this version.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

muyir123, 2024-11-15:
We need to cooperate with the new machine firmware, which has not been pushed yet

JohnSmithCYM, 2024-11-18:
Like what @muyir123 said the firmware has not been pushed and maybe join Firmware Beta Project in Handy could help.

closed 2024-12-09 by zhimin-zeng-bambulab