bambulab/BambuStudio #5289: Ghost K-Factor offset (Flow Dynamics Calibration) in new release.

MooNSparky, 2024-11-15:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Series 

### How to reproduce

1.1 Use auto calibration with the start of the print
1.2 Use previously used K-Factor settings which were working fine before
2. Used K-Factor from Par. 1.1 or 1.2 settings generate bad results
3. Add few points to the "Bad" K-Factor setting manually
4. Used K-Factor from Par. 3 settings generate good results

### Actual results

K-Factor using Lidar Auto Flow Calibration, 6 spools of the same PETG filament: 0.032-0.036 (0.035 chosen - just as before): the same result comes with selecting of Auto Calibration before printing 
[image]

### Expected results

Manual increase of K-Factor to 0.040 and tying it to spools in AMS
[image]

### Project file & Debug log uploads

No files are needed. Same results come from using previous settings or auto-calibration on my X1C and X1E. So, every old or auto recieved K-Factor is needed to be modified manually on my side. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-11-27:
@MooNSparky  If you perform automatic calibration again, can the new k value obtained meet the print quality requirements?

shuizhiguo, 2024-11-28:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

MooNSparky, 2024-12-18:
@zhimin-zeng-bambulab @shuizhiguo 

Hello!

I submited support ticket with logs and prints using X1E, number is: US241218707005

However, there seems to be another problem which appeared after I decided to factory reset my printers, just few days after I started this topic.

Instead of receiving better quality with manual K factor, I am getting the same result at any time: with auto (which shows k = 0.029) to manual (up to k = 0.045). Photos of the parts are also included into the support ticket. It seems that both printers don't want to change k factor from default at all. 

Thanks.