bambulab/BambuStudio #5291: Failed processing of file name format

Zoch707, 2024-11-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.2.1

### Printer

Creality Ender 3 V3 KE

### How to reproduce

slice file and try to export g-code. 

### Actual results

the little window pops up and says failed prossesing of file name format ...

### Expected results

it should ask me where to save the .gcode file and proceed to save it there.  

### Project file & Debug log uploads

[Auto_TestGit.3mf.zip](https://github.com/user-attachments/files/17777913/Auto_TestGit.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dcs3473, 2024-11-15:
Ditto.  This has happened to me also.  It has to do with this: Gcode Output: File Format:{input_filename_base}_{filament_type[initial_tool]}_{print_time}.gcode coming from the Orca Profile, I think.  It looks like it was copied over and has leftovers that are not supported in BambuStudio

XunZhangBambu, 2024-11-22:
@Zoch707 We did not include initial_tool as a placeholder internally. To avoid affecting your export, you can change the format to: {input_filename_base}_{filament_type[0]}_{print_time}.gcode, which will use the first selected material type as the material type in the exported file.

dcs3473, 2024-11-23:
Found it.  Need to turn on Dev Mode to see the missing field.  And that worked.  Thanks.

kkeonline, 2025-01-09:
Just try Bambu studio with Creality Ender 3 V3 KE and got the same error.
[image]
checking the json files all ender model are wrong.