bambulab/BambuStudio #5292: Issues adding new filaments

alvesjc, 2024-11-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.0.1

### Additional system information

2,6 GHz 6-Core Intel Core i7
Radeon Pro 555X 4 GB
Intel UHD Graphics 630 1536 MB
16 GB 2400 MHz DDR4

### Printer

X1C

### How to reproduce

1st create new filament and do all the calibrations with it.

Then go to Devices and select the new filament, color and PA profile. Hit confirm.
The selected value appears, flashes, and then disapear again.

Not sure if it's because of this, but then it doesn't get synced with printer, so can't also select it in printer.

### Actual results

the AMS spool returns to "?" state

### Expected results

The selected filament should be kept.

### Project file & Debug log uploads

[porta+etichetta+dry+box+profile.3mf.zip](https://github.com/user-attachments/files/17778933/porta%2Betichetta%2Bdry%2Bbox%2Bprofile.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-12-02:
@alvesjc What is your X1C firmware version? And can you reproduce this bug now?

alvesjc, 2024-12-02:
> @alvesjc What is your X1C firmware version? And can you reproduce this bug now?

Hi @JohnSmithCYM ,

1.8.2 version.
I assume I can, Studio was not yet updated and printer also same FW.
Also for your reference, the workaround I found to add the filament properly was to do it with old BambuStudio, version 1.9.7.52.
In this version it synced normally, and then I can use it in the new BambuStudio without issues.

I'm in a printing spree, but I can try to test it again with some filament to see if it changed when I've the printer finished. Let me know if it's required.

Thanks