bambulab/BambuStudio #5295: calibration not working

CookieCrypt, 2024-11-15:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

mac os

### Printer

bambu lab p1s combo

### How to reproduce

just try to do calibrate

### Actual results

doesnt allow you to calibrate 

if i go back 2 versions it still works but the new super tack build plate option is not there

### Expected results

should allow me to calibrate

### Project file & Debug log uploads

its just the calibration

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-18:
@CookieCrypt I have tried my P1S with dynamic calibration and flow calibration and both were right with the new super tack.

sablue1113, 2024-11-22:
Same problem

zhimin-zeng-bambulab, 2024-11-27:
@CookieCrypt  Could you take a screenshot to show that you are not allowed to calibrate?

Panda631, 2024-12-15:
I running BBS with win11, but faced with the same problem. I'm not allowed to calibrate

[image]

maruel, 2024-12-20:
Dupe of #5474

zhimin-zeng-bambulab, 2025-02-05:
It has been fixed in v1.10.2 Public Beta

closed 2025-02-05 by zhimin-zeng-bambulab