bambulab/BambuStudio #5297: Settings keep changing by themselves

i00, 2024-11-15:
### Bambu Studio Version

1.10.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Open a project
2. Preview the print
3. Close the project

### Actual results

A window is presented upon step 3 asking me to save changes - these changes were made automatically
[image]

### Expected results

Settings should not be changed automatically

### Project file & Debug log uploads

[Cat+Feeder+v1.zip](https://github.com/user-attachments/files/17782514/Cat%2BFeeder%2Bv1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

JohnSmithCYM, 2024-11-18:
Yes, I believe what @LJKraus explained is right.

LiZ-Li-BBL, 2024-11-28:
These changes you were asked were already made when you imported a 3mf.