bambulab/BambuStudio #5303: Object stacking broken after v01.09.07.52

dremugit, 2024-11-16:
### Bambu Studio Version

v01.10.00.89 and v01.10.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1 Carbon

### How to reproduce

Open shelftag plabasic base.stl, click Objects (vs Global), right-click shelftag plabasic base.stl, click Add part, Load, shelftag plabasic text.stl

### Actual results

On v01.10.00.89+, the two parts sit on the build plate and don't stack.

### Expected results

On v01.09.07.52, the two parts stack tidily, text on top of base and centered (well, slightly offset by design for the mounting holes.) 

### Project file & Debug log uploads

[workingandbrokenprojects.zip](https://github.com/user-attachments/files/17786900/workingandbrokenprojects.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dremugit, 2024-11-16:
[stl files.zip](https://github.com/user-attachments/files/17787161/stl.files.zip)

Oops, don't know if these are in the project files or not, so added these separately.

Haidiye00, 2024-11-18:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

dremugit, 2024-11-18:
[gifs.zip](https://github.com/user-attachments/files/17798253/gifs.zip)

One each for v01.10.01.50 ("broken") and v01.09.07.52 ("working")

ClaudioVarandas, 2024-11-18:
can confirm, object stacking by adding part is not working anymore, it adds the adds the part aside the parent object.

JohnSmithCYM, 2024-11-29:
Confirm,it will be fixed in next Github Release version.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00