bambulab/BambuStudio #5304: Opening STEP files takes a long time V1.10.1

Empire4191, 2024-11-16:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

Intel CPU

### Printer

X1C

### How to reproduce

Open STEP File

### Actual results

Opening STEP files takes a long time

### Expected results

No response

### Project file & Debug log uploads

Opening STEP files takes a long time

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DearTanker, 2024-11-27:
@MackBambu Are there any recommended import parameter values ​​for the P1S model with a 0.4 nozzle? For example, different levels such as high quality, balanced, default, etc. and their corresponding parameter values?

MackBambu, 2024-11-28:
> @MackBambu Are there any recommended import parameter values ​​for the P1S model with a 0.4 nozzle? For example, different levels such as high quality, balanced, default, etc. and their corresponding parameter values?

There are no specific recommended values for different machine models. This parameter controls the resolution when converting a STEP file into a triangulated mesh model. If you find the imported STEP file surfaces are not smooth enough, you can try reducing these two parameters to achieve a higher quality model.

DearTanker, 2024-11-28:
@MackBambu Both parameters can be set to very small values, but each adjustment takes a very long time to calculate (with freezing), so it takes a long time to wait and observe, which brings a very bad experience. At least a detailed introduction to the difference between the two parameters and a rough range of values ​​through WiKi would be very helpful.

MackBambu, 2024-11-29:
> @MackBambu Both parameters can be set to very small values, but each adjustment takes a very long time to calculate (with freezing), so it takes a long time to wait and observe, which brings a very [...]

@DearTanker Are you using macOS? Currently, the freezing issue has only been observed on macOS, but it's difficult for me to reproduce it on Windows.

DearTanker, 2024-11-30:
@MackBambu Low parameters make windows freezing all the time. my spec is i7 13700K 48GB 3080Ti. CPU loading 100% long time.

DearTanker, 2024-12-03:
[image]
[image]

Any slightly complex STEP file can freeze the UI, during which time other Windows system interfaces will also be stuck.

[DearTanker AMS Riser V2 Duble Cricle Dock.zip](https://github.com/user-attachments/files/17993621/DearTanker.AMS.Riser.V2.Duble.Cricle.Dock.zip)

MackBambu, 2024-12-03:
@DearTanker I'm using an i7-12700F with 32GB of RAM, but couldn't reproduce the issue. I need to test it on a 13th-gen Intel CPU.
https://github.com/user-attachments/assets/369e4267-d559-4b74-a267-fbdfb8ffcf5c

MackBambu, 2025-04-29:
The new version has already switched to a different threading model. Does the issue still persist?

Empire4191, 2025-04-29:
It has been working again for some time

closed 2025-04-29 by MackBambu