bambulab/BambuStudio #5305: Can't use "Replace with STL..." with STEP file since 1.10

macwhiz, 2024-11-16:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.0.1 (24A348)

### Additional system information

Mac Mini 2023, M2 chip, 16 GB RAM

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Create a new project
2. Import a STEP file (I used a file exported from Onshape as a version AP203 STEP file)
3. Slice and print the file (works normally)
4. Modify the original object in CAD and export a new STEP file using the same export settings
5. Select the object in Bambu Studio, right click, and select "Replace with STL"
6. Select the updated STEP file

### Actual results

Nothing happens. The object isn't replaced. The new STEP import dialog doesn't appear. To replace the object, one must delete it and re-import it, as this function no longer works with STEP files. (Prior to the introduction of the new STEP import dialog, it worked with STEP files as well as STL files.)

### Expected results

The STEP file should load and replace the selected object. The new STEP import dialog may appear.

### Project file & Debug log uploads

[Eureka PowerSpeed - Top Hose Cap for debug.3mf.zip](https://github.com/user-attachments/files/17787615/Eureka.PowerSpeed.-.Top.Hose.Cap.for.debug.3mf.zip)
[debug_Thu_Nov_14_12_11_51_58046.log.0.zip](https://github.com/user-attachments/files/17787618/debug_Thu_Nov_14_12_11_51_58046.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-11-18:
The issue has been fixed. Please test it.
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

Sweatyhogg, 2024-11-29:
@MackBambu I've tested this on the linked v01.10.01.50 and it didn't work. Went back to previous 1.09 and works again.

kedumini, 2024-12-06:
I also cannot replace a parameterized STL object with a STEP using the "replace with STL" option in v1.10.1

MackBambu, 2024-12-12:
> @MackBambu I've tested this on the linked v01.10.01.50 and it didn't work. Went back to previous 1.09 and works again.
> 
> Same details as original issue, apart from Windows 11.

You are right .This is another bug, and now it can be confirmed.

TwoRats55, 2025-02-02:
> [@MackBambu](https://github.com/MackBambu) I've tested this on the linked v01.10.01.50 and it didn't work. Went back to previous 1.09 and works again.
> 
> Same details as original issue, apart from Windows 11.

I'm having this issue also (BS v1.10.1.50 on Windows 11).

ddmgh, 2025-02-06:
Can confirm this broken behavior with the 1.10.1.50 Flatpak on Linux.

philipbergstrom, 2025-02-10:
Same issue here. V. 1.10.1.50. 
Windows 10

phiberhippo, 2025-02-10:
Same issue here. V. 1.10.1.50.
Windows 10

ddmgh, 2025-02-25:
Can confirm this broken behavior with the 1.10.2.75 Flatpak on Linux.

TwoRats55, 2025-02-25:
> > [@MackBambu](https://github.com/MackBambu) I've tested this on the linked v01.10.01.50 and it didn't work. Went back to previous 1.09 and works again.
> > Same details as original issue, apart from Windows 11.
> 
> I'm having this issue also (BS v1.10.1.50 on Windows 11).

Still broken in latest update v1.10.2.76 on Win11.

MackBambu, 2025-03-26:
Fixed in v2.0

closed 2025-03-26 by MackBambu

ddmgh, 2025-08-24:
This issue appears to be back and I'm experiencing it with 2.2.0.85.

TwoRats55, 2025-09-10:
> This issue appears to be back and I'm experiencing it with 2.2.0.85.

Appears to be working in 2.2.1.60.