bambulab/BambuStudio #5307: v1.10.1 Support Interface pattern bug - Rectilinear Interlaced and Rectilinear identical

tommyboy180, 2024-11-16:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

CPU: Ryzen 5800x3d
RAM: 32GB 
GPU: 3080 Nvidia

### Printer

X1C

### How to reproduce

Set Raft layers to 1 or more. Interface pattern default selected, but the slicer will generate "Rectilinear Interlaced". Selecting "Rectilinear" or "Rectilinear Interlaced" results in the same pattern. 

### Actual results

Interface pattern defaults to "Rectilinear Interlaced" even when set to Default or "Rectilinear", giving support interface interlaced. 
[image]
[image]

### Expected results

Selecting "Rectilinear" should be support interface lines in parallel at the angle specified in "pattern angle"
[image]

### Project file & Debug log uploads

[Interface support bug.zip](https://github.com/user-attachments/files/17787826/Interface.support.bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tommyboy180, 2024-11-16:
Version 1.10.1 Interface pattern defaults to "Rectilinear Interlaced"when set to "Default" or "Rectilinear". 

Bug - The option "Rectilinear Interlaced" and "Rectilinear" are identical.

tangketan, 2024-11-22:
Yes, we've found this issue and will fix it in the next version. Thanks for reporting.