bambulab/BambuStudio #5308: Can no longer select both Surface and Horizontal text in the edit text menu

SK360, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Add Text
2. Click Horizontal Text in the edit text window, Surface becomes unchecked.

### Actual results

[image]

### Expected results

It used to be possible to select both, shown here is from 1.8.xx

[image]

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-18:
Thank you for discovering the issue. This feature is currently being fixed~

Watio2472000, 2024-11-19:
Thanks for this. I am having this issue too.

tedcoffelt, 2024-11-19:
I need to have this feature back for an immediate use.

When will this be repaired?

Is there a way for me to move backward on the OS?

Ted

Haidiye00, 2024-11-20:
Hello,we are currently repairing it~.At present, you can import with the old version first, and then continue editing with the new version. Sorry for the inconvenience caused to you~

tedcoffelt, 2024-11-20:
Thank you for the suggestion about loading an old version of the OS.

QUESTION - can you please provide a link to an old version?  I was not able
to locate one on the website.

Ted

Haidiye00, 2024-11-21:
> 感谢您提供有关加载旧版本操作系统的建议。 问题 - 您能提供旧版本的链接吗？我无法 在网站上找到一个。 泰德
> […](#)

https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00