bambulab/BambuStudio #5311: AMS Loaded filament not shown

PLSomebody, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.1 (24B83)

### Additional system information

MacBook Pro 16 M2Max

### Printer

X1C

### How to reproduce

Start any project using filament from AMS

### Actual results

Presently used filament should be shown in AMS diagram.
Attached screenshot shows the problem. 
Arrow shows that printer uses white filament (which is in bay 2 of AMS) and red box shows that AMS diagram doesn't show which filament is used.
[image]

### Expected results

As in previous Bambu Lab Studio versions, app should shown which filament is loaded by making it bold

### Project file & Debug log uploads

[Log.zip](https://github.com/user-attachments/files/17789422/Log.zip)
[Project DryBoxSet.3mf.zip](https://github.com/user-attachments/files/17789423/Project.DryBoxSet.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

PLSomebody, 2024-11-17:
It seems that above problem is due to lack of filament diagram update. When you will switch to Preview tab and then back to Device tab, diagram shows loaded filament as you can see on below screenshot:
[image]

You can see one other problem with display: layer number and remaining time is broken (indicated by red box on above screenshot)

walterwongbbl, 2024-11-29:
@PLSomebody 
This seems to be caused by insufficient page refresh times, we will solve it.