bambulab/BambuStudio #5312: Don't disconnect printer in LAN mode

UBaer21, 2024-11-17:
**Is your feature request related to a problem? Please describe.**
When switching tabs the "Device" tab often shows "no printer" as it disconnects.
Also don't stop the camera (if liveview when printing is on) when switching tabs.
I often need to check something and when switching back the video is off and sometimes also the printer disconnected 
(the disconnect doesn't happen in the previous version 1.9  before 1.10.1.50)

**Describe the solution you'd like**
choosing my printer again and again and again from the dropdown

**Describe alternatives you've considered**
stop using BambuStudio

DarkSwoop, 2024-11-17:
Same thing here. Really annoying.

UBaer21, 2024-11-17:
@DarkSwoop yeah but bug reports are so annoying to fill out.

DarkSwoop, 2024-11-18:
https://github.com/bambulab/BambuStudio/issues/5315 may be related to this.

miikememe, 2025-01-20:
Still a bug, macOS bambu v 1.10.0.89

nateabele, 2025-03-27:
Yeah, this is definitely a bug. I have a P1S and this affects all releases of BambuStudio for the past year+.

It disconnects. **CON**. **STANT**. **LY**.

It disconnects if I switch tabs. It disconnects if I minimize the window. It even disconnects if the window sits idle or is in the background for too long.

Given all the other signals (in software releases, on the forums, and elsewhere) indicating how antagonistic Bambu is towards LAN mode and their incentives towards driving people to their cloud service, I can only assume this is intentional.

If anyone's able to make a patch, and I'm happy to fork it and build / release a working version.

mkstephenson, 2025-06-18:
Yeah, this has been bugging me for a long time. I would be happy with an option to auto-reconnect the previously selected printer when switching to the device tab. I imagine the connection is auto dropped to save on resources (as annoying as that is).