bambulab/BambuStudio #5313: The lines set by AMS will disappear.

yuang093, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10 22H2

### Additional system information

裝置名稱	DESKTOP-MNT1QSU
處理器	Intel(R) Core(TM) i5-8600K CPU @ 3.60GHz   3.60 GHz
已安裝記憶體(RAM)	32.0 GB
裝置識別碼	309A81D2-15A4-4754-A8DA-F07213521B63
產品識別碼	00328-10000-00001-AA714
系統類型	64 位元作業系統，x64 型處理器
手寫筆與觸控	此顯示器不提供手寫筆或觸控式輸入功能

### Printer

Bambu X1C

### How to reproduce

1. 設備
2. AMS
3.  設定線材廠牌顏色
4. 設定跑掉需要重新設定
[image]

### Actual results

[image]
在機器上面設定好線材廠牌與顏色後
設定會消失

### Expected results

線材設定會自動消失

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17790484/log.zip)
[log.zip](https://github.com/user-attachments/files/17790509/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

JohnSmithCYM, 2024-12-02:
@yuang093 Can your reproduce this problem? And can you offer your X1C firmware version for us to get a further test?

yuang093, 2024-12-02:
目前已經正常
It's normal now