bambulab/BambuStudio #5314: AMS Sync issue in Bambu Studio

parmarneil, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 latest release 24H2

### Printer

P1S

### How to reproduce

Goto Prepare & Device. Mismatch of info on he AMS,

### Actual results

I have an AMS with only 3 x filament reels installed, slot 1, 2 & 4 only.
BS shows slot 1, 2 & 3 used. Regardless of sync it does not match the info shown on the device tab. This was all working fine until BS was updated to the latest release.
[image]
[image]

### Expected results

You can see the mismatch in info. Slot 3 is empty which is correct.

### Project file & Debug log uploads

[NOCO BOOST PLUS TOP.zip](https://github.com/user-attachments/files/17791010/NOCO.BOOST.PLUS.TOP.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

parmarneil, 2024-11-17:
Just thought I'd let you know of this bug.

JohnSmithCYM, 2024-11-18:
You mean the old version studio will show filaments 1,2,4 to Black? I guess it will show 1,2,3 even in old version due to you can't jump a number in filament setting not only in newest version or old ones.

Rocketzilla, 2024-11-21:
Yeah, I'm going bananas too.  I'm having trouble getting into my Bambu Lab X1c.  This machine has been banging on all cylinders for about a year without any major problems. It's the software that's been giving me grief big time. When I try to run a job it fails to see my X1c. Not a big deal right? Flush the email address and password then re-enter. Noop.. I get a password error. I try it a few times to make sure it's rightest. No luck on that one either. Okay, re-entered - but this time hit the "send new password" option. I'm waiting, twenty minutes go by. An hour and a half goes by. Okay, I know it wasn't a grate idea, but a desperate man does desperate things. I re-entered and just for good measure, I re-entered a third time. That was yesterday afternoon. It's been about 24 hours now and not a peep from Bambu Lab. I know they're like 3,000 miles away, but common... This new issue has problems. Great. So what now? I could just put the project on a card and run it direct. But I have multiple plates to run and with out the AMS, that will make life difficult. I slap myself for not saving the old issue, I'll tell you what. When Bambu Lab gets around to fixing this I'm gonna expect a kiss - and you know where.

JohnSmithCYM, 2024-11-22:
[image]
@parmarneil I don't know if it is what you want: you can click any filament to get a list of ams filaments, it shows AMS (which was represented in A,B... as you get more that 1 AMS) and filament slot information, I hope that will help you.

JohnSmithCYM, 2024-11-22:
@Rocketzilla It seems you have encountered a problem of Cloud service of Bambu, have you tried to contact them by handy?Or you can leave the e-mail here and maybe Bambu employee will see that. @ttt888zzz

Rocketzilla, 2024-11-23:
Thanks for getting back to me John. I wasn't sure how to do that, since I wasn't aware they have a direct line of contact by email. I've been banging away with the problem of getting Bambu Studio to recognize my printer for a week now; going through all the machinations of entering the serial number, IP address, access code, and whatnot - but still no dice. It pukes on the login and doesn't allow me the option to fix it. Maybe I'll have better luck soon.
 
Michael

JohnSmithCYM, 2024-11-25:
@Rocketzilla You could see https://wiki.bambulab.com/en/create-a-support-ticket to know how to create a support ticket, I hope this could help you.