bambulab/BambuStudio #5317: Support don't obey threshold angle

UBaer21, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S

### How to reproduce

add sphere activate support

### Actual results

[image]

### Expected results

more support

normal Support expansion = 0.1 can fix that but is 0 by default

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-11-18:
I couldn't reproduce the issue. Would you please upload the 3mf?
[image]

UBaer21, 2024-11-18:
This seems to be an issue of layer height (adaptive layer)
see here (default bambo print process 0.6 nozzle) support set for 90° which should be half the sphere
0.05
[image]
  0.2
[image]

0.5
[image]

changing layer height should not change the support angle/height

NothFie, 2024-11-28:
Thank you for your reporting. You are right that the change of the layer height does affect the generation of support. Because with the reduction of the layer heights, changes in the area of the upper and lower layers also reduce. Then we consider these small overhangs support-free. If you need the support to cover half of the sphere, please tell us your situation and we will try to meet your needs.

UBaer21, 2024-11-28:
@NothFie  shouldn't the "remove small overhangs" have any effect on this? Because it doesn't.